leggedrobotics/raisimLib #1: A few questions

maximecb, 2019-07-22:
Hello. Great work on the physics simulator. Happy to see an alternative to PyBullet pop up. I have a few questions:

1. Do you have, or plan to have Python bindings for your physics simulator?

2. Is the simulator in a state where one could easily implement a gripper arm? Any reason why this would not work?

3. Can you easily support multiple simulator instances in one process?

4. Does your simulator support polygonal meshes? Could I pass a list of triangles/faces to it programmatically? (without loading from a model file)

jhwangbo, 2019-07-22:
Are you planning to do RL with raisim? then please take a look at our gym examples (https://github.com/leggedrobotics/raisimGym)

1. We have a gym example of raisim which wraps C++ at the environment level using pybind11. This is how I do RL since it takes very little time to compile (5~10 secs) and still very efficient since the parallelization is done at C++ level. Try raisimGym and let me know if you find a better way.

2. There is a kinova example in raisimOgre. Students in our lab are using raisim for manipulation projects

3. Yes, check out raisimGym examples. raisim::World doesn't have a copy constructor. So I just make a std::vector<std::unique_ptr<raisim::World>> to manage multiple instances

4. This will be done in the near future but not yet. I use ODE for collision detection but it seems to have a few memory issues. I implemented/reworked heightmap since that's what I need now. I can delete and recreate a heightmap programmatically. It will be the same for polygonal meshes.

maximecb, 2019-07-22:
Out of curiosity, what is `RaisimServer`? (this class could use a comment explaining what it's for)

https://github.com/leggedrobotics/raisimGym/blob/master/raisim_gym/env/RaisimServer.hpp

I'm assuming this is to run a simulation on a different process than the training? Can you run everything on the same machine without networking?

jhwangbo, 2019-07-22:
Simulation runs on a single machine. It's up to you to choose which machine to be used for visualization. RaisimOgre is meant to be used in the same process as the simulation. Check out our raisimOgre examples. But you can use RaisimServer to do remote visualization. Frankly, it is not working well now. It is on my to-do list.

closed 2019-07-24 by jhwangbo

bdelhaisse, 2019-08-06:
Hey guys,

>     1. Do you have, or plan to have Python bindings for your physics simulator?

I recently wrote the Python wrappers for `raisimLib` and `raisimOgre` using `pybind11`. Most of the code is wrapped except few stuffs (e.g. methods from Ogre or the Raisim server). Anyway, you can already find it on the following link:
https://github.com/robotlearn/raisimpy

For now, I provided several python examples that are based on the C++ examples that can be found in the `raisimLib/examples/` and `raisimOgre/examples/` folders. I also added a part of the RL examples that are in `raisimGym` (the `anymal_ppo3.py` one has a bug but all the other examples should normally work).

Also, note that I didn't test all the methods yet, and there are probably few stuffs that could be improved with the wrappers; it was my first time using `pybind11` so I learned while coding the wrappers, and realized later I could improve few previous parts.

I also added the documentation for each method but some parts were obscure and I had to infer what the methods were doing, or what their arguments stand for. I also found few typos in the original code, but I will open another issue for that, or submit a pull request.

Best,
Brian

jhwangbo, 2019-08-06:
Thx Brian, The code and API seem very clean! I will try it out this week and addresses some of the issues you listed in the readme. I'll also add a link to it on all readme's of raisim repos.