leggedrobotics/raisimLib #3: wrong getGeneralizedForce()

floodsung, 2019-08-20:
I tried to print forces of anymal:
std::cout << anymal_->getGeneralizedForce()<< std::endl;

below is what i obtained from the terminal:
0 0 0 0 0 0 0 -19.0064 16.9814 5.15152 35.6953 -3.19008 18.4637 6.1331 3.68948 -22.224 -32.2061 0.290263 
0 0 0 0 0 0 0 -7.06745 -0.517137 22.6091 -8.31036 10.9473 5.02551 -11.7917 -8.03127 -15.1463 24.6648 -13.6599 
0 0 0 0 0 0 0 -0.59681 19.526 12.1407 1.15952 0.182062 14.3934 -34.4135 -10.8773 16.3055 -21.6263 -24.984 
0 0 0 0 0 0 0 17.7714 1.16777 -14.7985 2.69269 3.34732 -9.81864 -3.19977 4.29663 -4.16517 -22.4896 -12.2793 
0 0 0 0 0 0 0 14.8251 3.44768 14.497 1.16175 2.67442 28.0617 5.82688 6.39095 -14.8618 12.9979 12.1853 
0 0 0 0 0 0 0 -36.7177 -2.99206 10.8263 -2.13563 15.5825 8.95963 18.5383 -14.7241 25.108 -14.4022 -14.4026 
0 0 0 0 0 0 0 3.41373 5.44599 -22.2235 17.8641 -12.9422 3.76054 5.64001 6.73451 -8.9726 25.9793 -21.7382 
0 0 0 0 0 0 0 -1.22766 -6.14286 -14.1164 -7.7704 12.8151 -24.2552 -11.0073 -8.60783 1.67216 10.5513 -8.63154 
0 0 0 0 0 0 0 -13.5507 -1.41343 20.1651 12.0834 5.55141 -15.6579 1.21728 12.7228 -25.7639 12.8844 -16.4923 
0 0 0 0 0 0 0 -51.6053 16.3171 13.2201 -13.1371 -3.9688 12.5896 -0.45729 6.88962 11.1185 -12.0802 5.0165 
0 0 0 0 0 0 0 -33.0467 55.8757 -3.79658 12.9501 -2.82743 19.9766 -2.42161 -18.7575 9.14279 37.1704 -1.18315 
0 0 0 0 0 0 0 -31.9512 53.2185 17.4517 -5.94892 -3.97466 2.78641 -47.3691 -8.12961 9.72698 24.1693 -21.2707 
0 0 0 0 0 0 0 -57.0449 27.9103 24.4798 -30.7587 -5.63999 -15.3937 -37.8111 -30.2737 -32.9886 24.7853 5.4005 
0 0 0 0 0 0 0 -19.4207 6.2203 -13.9687 0.240711 2.30872 -4.20359 2.66173 -10.2754 19.9118 -1.92223 0.735265 
0 0 0 0 0 0 0 2.09981 -6.85247 22.7617 -5.69461 10.0011 1.24108 -10.9685 5.78047 6.15456 -4.63611 -15.5814 
0 0 0 0 0 0 0 17.4764 7.43938 7.22429 11.2891 -14.4075 -2.88991 20.5445 3.63869 -34.4899 -9.58554 -16.8402 
0 0 0 0 0 0 0 -19.5238 -13.2854 9.57525 7.21913 3.24537 0.732437 -3.73253 -14.5557 -20.8852 -6.36125 -10.4515 
0 0 0 0 0 0 0 -1.79329 -1.78065 -7.42062 2.08612 26.5753 3.16248 -20.1832 12.2045 3.36962 10.2394 7.82083 

there should be 12 torques but we have the output of 18 dimension,and only 11 elements have real value. How to explain this? thank you!

jhwangbo, 2019-08-20:
Hi floodsong, generalized force should have the same dimension as the generalized velocity (a.k.a. DOF). So it should be 18-dimensional. The first 3 are x,y,z degrees of freedom of the main body. if they are set to zero, it means that the body is not externally pushed in any direction. The next three are degrees of freedom associated with the angular velocity. if they are set to zero, it means that there is no external torque acting on the body.

I just checked with raisimOgre and it prints out the correct numbers. In your case, I am not sure why 7th element is always zero. You probably set them yourself. One possible confusion is from the difference between the generalized coordinate (GC) and the generalized velocity (GV). Since the orientation is represented by a quaternion, the first 7 elements of the GC are associated with the main body. The first 6 elements of the GV are associated with the main body. If this confused you, check raisim manual (https://slides.com/jeminhwangbo/raisim-manual#/articulatedsystem). 

If can also be due to the version you are using. What is the version number?

floodsung, 2019-08-20:
I have the same right result in raisimOgre. 

Could you check it in raisimGym?

floodsung, 2019-08-20:
I use the latest version

jhwangbo, 2019-08-20:
this is actually a bug in the raisimGym example. I am making a fix now. Thank you for reporting!

jhwangbo, 2019-08-20:
Hi Sung, can you pull all repos and try again?

jhwangbo, 2019-08-20:
Basically, the issue was that the P gain had 19 dimension instead of 18. This set one of the p gains to zero (one of the abduction joint). It should be fixed now

jhwangbo, 2019-08-20:
closing the issue. This was a bug in raisimGym. from raisim v0.3.1, setPDgain() checks the dimension automatically

closed 2019-08-20 by jhwangbo

floodsung, 2019-08-21:
the bug is fixed ! awesome!