leggedrobotics/raisimLib #5: unable to use getCollisionObj()

floodsung, 2019-08-23:
hi, I tried getCollisionObj() but compiled error:

error: ‘raisim::CollisionSet& raisim::ArticulatedSystem::getCollisionObj()’ is protected within this context

how to deal with this problem?
Thx

jhwangbo, 2019-08-23:
getCollisionBodies() is the public method (doing the same thing). Your IDE should hide the other one. How did you find about that method??

floodsung, 2019-08-23:
I found this method in your science paper code. I am trying to obtain the foot contact position.
is it possible in current version?

jhwangbo, 2019-08-23:
The API changed. SR version is very old and the API was not very stable then. I don't guarantee that my old code runs here. But of course no functionalities are lost and you can do the same with the new API. The collision bodies should give you a non-const list of collision bodies. You can do whatever you want to do with it. 
If you are trying to reproduce the work and modifying the mass, don't forget to call updateMassInfo() which will keep necessary internal variables up to date.

closed 2019-08-23 by jhwangbo

floodsung, 2019-08-23:
Thank you very much.

One more question: what is the functionality of getVisColProps() in your old code? is it the same with the new API: getVisColOb() or getVisColObPose()? thx

jhwangbo, 2019-08-23:
it should be the same as getVisColOb()

floodsung, 2019-08-23:
Ok, Thank you so much