leggedrobotics/raisimLib #6: Collision body could be stuck in the ground

floodsung, 2019-08-26:
[image]
I tried to calculate foot positions but found that collision bodies could be stuck in the ground. As shown in the above picture, the height of foot is negative.

is it a bug?

floodsung, 2019-08-26:
I tried a initialization of anymal with height 1.0.

jhwangbo, 2019-08-26:
This is due to the numerical integration. RAISIM doesn't perform continuous collision checking (as far as I know, none of the major simulators do this). If the foot has high velocity, it can penetrate into the ground. This can be mitigated by having a smaller time step.

Can you further explain what you are simulating? Do you use a pd controller and set constant tartget positions?

floodsung, 2019-08-26:
I am just using your sample anymal code, but printing the food position. It use the same pd controller as you used in the code. I am wondering whether or not this problem would affect the replication of your science paper.

floodsung, 2019-08-26:
what i printed is the initial state of the anymal's foot.

jhwangbo, 2019-08-26:
I used the same setting in the science robotics paper. You can go for a smaller time step if you don't want to see such an artifact. but this, in my oppinion, should make the trained controller more robust.

floodsung, 2019-08-26:
ok, thx.

closed 2019-08-26 by floodsung

jhwangbo, 2019-09-27:
Hi Floodsung, I found a way to improve on this. I just pushed my changes. If the penetration was an issue, you can pull and get the latest version