leggedrobotics/raisimLib #8: Change gravity vector?

rmbp, 2019-09-05:
Hi Jemin,

Thank you very much for releasing this, such a useful tool.

I can't figure out how to change the gravitational acceleration... Have tried directly defining the vector in World.hpp in the include directory of LOCAL_BUILD, as well as changing stuff in ode, but nothing seems to be changing the behaviour of the robot when I build and launch my environment in raisimGym...

Am I missing something?

Thanks.

jhwangbo, 2019-09-05:
You can change the gravitational acceleration by ```World::setGravity()``` method. I just checked that the method and it works fine but let me know if you have any issue. Please don't change the header file unless you really know what you are doing. It can break the class since you are working with the compiled binaries. ODE only performs collision detection and you don't have to touch it in raisim.

closed 2019-09-05 by jhwangbo

rmbp, 2019-09-05:
Ah perfect thanks very much.

William Jones

PhD Student
Space Robotics Laboratory, Tohoku University