leggedrobotics/raisimLib #10: Question about 'updateKinematics'

dytan, 2019-09-10:
Hi, little confused about the function [ArticulatedSystem::updateKinematics()](https://github.com/leggedrobotics/raisimLib/blob/master/include/raisim/object/ArticulatedSystem/ArticulatedSystem.hpp#L103). 
    
What operation it will do if call this manually ? For the situation only want the kinematics without run simulation, after call setState, 'getFramePosition' return the updated pose but 'getFrameVelocity' not change, and call 'updateKinematics' makes no difference, instead 'preContactSolverUpdate1' can update the world vel of body and frame.

jhwangbo, 2019-09-10:
Good question. Actually you dont have to call it. setState and other methods that change the configuration will call updatekinematics automatically. It seems safe to make this method private but ill double check and then make it reflected on the next release

closed 2019-09-10 by jhwangbo

dytan, 2019-09-10:
Thanks very much. But still want to know is there a way to get frame or body world velocity without run simulation? For now, I have to call pre_contact_solver_update1(GRAVITY, 0), after this getFrameVelocity will return the updated value.

jhwangbo, 2019-09-10:
Sorry, I missed the second part of the question. you are not simulating when you call preContactSolverUpdate1. The actual integration happens in integrate2(). 

All necessary velocities are computed as part of Recursive Newton Euler (RNE), which is called in preContactSolverUpdate1(). This method is not the fastest way to get velocities but the implementation is quite efficient.

dytan, 2019-09-10:
Got it. Thanks you.