leggedrobotics/raisimLib #15: jointState(18) has 18 datas,how to define it . why we define jointPgain.tail(12)

yonchien, 2019-10-11:
jointState(18) has 18 datas,how to define it . why we define jointPgain.tail(12) , how to define the others

jhwangbo, 2019-10-11:
yonchien, please write a complete question so that others can understand what you mean. I understand your question but other users cant. I guess you mean why anymal example has 18 dof and why we only define the last 12 elements in pd gains.

For the first question, please refer to the manual. It explains how a generalized coordinate is defined for articulated systems. We only define the last 12 elements of pd gains because they are joints. The first 6 are related to the floating body. And we dont actuate the body. For floating base systems, first 6 elements in p,d gains are always ignored.

yonchien, 2019-10-11:
the manual explains a generalized coordinate position and quaternion , it has 7 elemnets not 6
[image]

jhwangbo, 2019-10-11:
Dof is not equal to the dimension of the generalized coordinate. Anymal has 18 dofs, and a 19-dimensional generalized coordinate

yonchien, 2019-10-11:
what the jointNominalConfig means?

jhwangbo, 2019-10-11:
nominal configuration of the joints for ANYmal. That's the generalized coordinate of ANYmal when we turn on ANYmal and the configuration where we start our simulation.