leggedrobotics/raisimLib #18: setExternalTorque() not implemented yet ?

Atlis, 2019-10-11:
Hello,

First of all, thanks for this awesome simulator!

I've been attempting to set an external torque to a body by using the setExternalTorque() function. Unfortunately, it returns "not implemented yet". Is there a simple workaround? Or, should I simply wait for the proper implementation, and if so, when do you expect it to be implemented.

Thanks!

jhwangbo, 2019-10-11:
I just implemented the method and pushed the changes. Please check and let me know how it works. I tested it on the ANYmal model.

Atlis, 2019-10-14:
Thank you very much for taking the time to implement the method. I have not tested it yet, but I can already tell you one of raisimOgre examples does not compile anymore. More specifically: the file deserializer.hpp because of the changes made related to raisim::RaisimServer::RAISIM_PORT.
Thanks again!

jhwangbo, 2019-10-14:
fix pushed ;)

closed 2019-10-15 by jhwangbo