leggedrobotics/raisimLib #19: can not add a cone

yonchien, 2019-10-12:
I just want to add cone . First ,  I define auto cone =world.addCone(0.1,0.1,10), cone->setPosition(1,2,0); Then ,  vis->createGraphicalObject(cone,"cone","default");
but it error " no matching function for call to ‘raisim::OgreVis::createGraphicalObject(raisim::Cone*&, const char [5], const char [8])"

and I can't found this function ,how can i  add a cone or object like triangle

jhwangbo, 2019-10-12:
cone collision body is not supported anymore, unfortunately. I'll delete the API in the next release.

The only way to add a collision body that's not a standard shape is by mesh.

closed 2019-10-12 by jhwangbo

yonchien, 2019-10-12:
what are your standard shape  included?

jhwangbo, 2019-10-12:
anything you can add by "world.add ..."