leggedrobotics/raisimLib #22: help on urdf file#2

alexpostnikov, 2019-10-16:
Hello, thanks for the work.

I want to simulate a [shunk gripper](https://schunk.com/us_en/gripping-systems/highlights/svh/) using raisim, but encounter a strange behavior. 
This gripper has a git repo with a ROS driver (there are available xarco urdf and meshes, Xarco urdf can be easily modified to pure urdf.). 

After loading the urdf if manualStepping is turned on, the 'setState' is ok, kinematic part seems ok. But when call 'world.integrate()' there happens **Segmentation fault**, *but not always just in half cases*. There are not accurate inertia parameters of links, but if the problem is in inertia, usually position of links would be expected as NaNs, not Segmentation fault errors

I've  tried to play with urdf params and it seems that after limiting of dof (from 20 to 3-4) errors disappears. Changes related to setTimeStep doesn't make any progress.

Were used just the current version of raisimLib master branch.

urdf file:
https://gist.github.com/alexpostnikov/3b14285a35ed0cc5f25aa079a6a81277

meshes can be found at shunk driver repo, or you can it (and urdf) from [gdrive](https://drive.google.com/drive/folders/1JHWAt0FClIkSIowPx51Wm-fH32G0uryG?usp=sharing)

[image]

jhwangbo, 2019-10-16:
I tried your URDF. This urdf will work fine if the hand is connected to a robot arm. But since the base is fixed, 5 fingers have their own independent dynamics. Technically, it is 5 different articulated systems, not a single one. 

When I first designed RaiSim, I excluded such a corner case since it will be faster to simulate them independently. But I guess the current version is not convenient. I'll push a change today so that you can run the simulation. I still cannot guarantee that all methods work with that system. If you have an issue, please report it here. You can also run it with valgrind to see exactly which method crashes. This will speed up my fix. 

Cheers

jhwangbo, 2019-10-16:
I pushed the change. Let me know how it works. Closing the issue for now

ra3vld, 2019-10-23:
Thx for fix bug now this urdf work in simulate in raisimOgre, but when I play with it in raisimGym with more than one thread it is crashing with seg fault on world_->integrate() with next stack trace.
```
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03b14c
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc59
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc59
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc42
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc59
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc59
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc42
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc42
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc42
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc42
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc59
Opcode::OBBCollider::_Collide(Opcode::AABBNoLeafNode const*) (.localalias.4) 0x00007fb8ba03bc42
Opcode::OBBCollider::Collide(Opcode::OBBCache&, IceMaths::OBB const&, Opcode::Model const&, IceMaths::Matrix4x4 const*, IceMaths::Matrix4x4 const*) 0x00007fb8ba03e575
dCollideCylinderTrimesh(dxGeom*, dxGeom*, int, dContactGeom*, int) 0x00007fb8ba00dc61
nearCallback(void*, dxGeom*, dxGeom*) 0x00007fb8bbde2ded
dxHashSpace::collide(void*, void (*)(void*, dxGeom*, dxGeom*)) 0x00007fb8b9fc9e6b
raisim::World::integrate1() 0x00007fb8bbde4dc7
raisim::World::integrate() 0x00007fb8bbde5859
raisim::ENVIRONMENT::step(Eigen::Ref<Eigen::Matrix<float, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> > const&) Environment.hpp:268
::_ZN6raisim21VectorizedEnvironmentINS_11ENVIRONMENTEE4stepERN5Eigen3RefINS3_6MatrixIfLin1ELin1ELi1ELin1ELin1EEELi0ENS3_11OuterStrideILin1EEEEESA_RNS4_INS5_IfLin1ELi1ELi0ELin1ELi1EEELi0ENS3_11InnerStrideILi1EEEEERNS4_INS5_IbLin1ELi1ELi0ELin1ELi1EEELi0ESD_EESA_._omp_fn.0() VectorizedEnvironment.hpp:186
__kmp_api_GOMP_parallel 0x00007fb91e5c8fc8
raisim::VectorizedEnvironment<raisim::ENVIRONMENT>::step(Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<bool, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&) VectorizedEnvironment.hpp:104
void pybind11::cpp_function::initialize<pybind11::cpp_function::cpp_function<void, raisim::VectorizedEnvironment<raisim::ENVIRONMENT>, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<bool, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, pybind11::name, pybind11::is_method, pybind11::sibling>(void (raisim::VectorizedEnvironment<raisim::ENVIRONMENT>::*)(Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<bool, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::'lambda'(raisim::VectorizedEnvironment<raisim::ENVIRONMENT>*, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<bool, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&), void, raisim::VectorizedEnvironment<raisim::ENVIRONMENT>*, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<bool, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, pybind11::name, pybind11::is_method, pybind11::sibling>(void&&, raisim::VectorizedEnvironment<raisim::ENVIRONMENT> (*)(Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&, Eigen::Ref<Eigen::Matrix<float, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<bool, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >&, Eigen::Ref<Eigen::Matrix<float, -1, -1, 1, -1, -1>, 0, Eigen::OuterStride<-1> >&), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::'lambda1'(pybind11::detai [... cut ...]
_PyEval_EvalFrameDefault 0x000055708a562ad0
_PyFunction_FastCallKeywords 0x000055708a50a79b
_PyEval_EvalFrameDefault 0x000055708a562846
_PyEval_EvalCodeWithName 0x000055708a4a84f9
PyEval_EvalCodeEx 0x000055708a4a93c4
PyEval_EvalCode 0x000055708a4a93ec
run_mod 0x000055708a5c1874
PyRun_FileExFlags 0x000055708a5cbb81
PyRun_SimpleFileExFlags 0x000055708a5cbd73
pymain_main 0x000055708a5cce5f
_Py_UnixMain 0x000055708a5ccf7c
__libc_start_main 0x00007fb92aac8b97
_start 0x000055708a572122
_start 0x000055708a572122
```

jhwangbo, 2019-10-23:
Are you testing with the same URDF that you sent me?

jhwangbo, 2019-10-25:
I ran your urdf in raisimOgre
[image]

The internal collision becomes very complicated in this model. Is this what you intended?

I think the main issue is the finger joint end. you can easily replace it by a joint limit.

closed 2019-10-26 by jhwangbo