leggedrobotics/raisimLib #23: Unknown body id with getBodyIdx

rmbp, 2019-10-18:
Hi,

When using getBodyIdx on a body attached to a fixed joint I get the error `ArticulatedSystem.cpp:1364] unknown body id`.

When changing the joint type to revolute, there are no issues. I have tested on my own URDF and that of the ANYmal example in raisimGym and they both produce the same problem.

Thanks

jhwangbo, 2019-10-18:
All fixed body properties are combined with their parents'. If you want to get a position and orientation of a specific part, you can get it at a joint. All joints are converted to Frames even if they are fixed ones. Please check tbe manual

By the urdf convention, a body might not have a positional property.

closed 2019-10-18 by jhwangbo

rmbp, 2019-10-18:
Ah yes, the manual.

Thanks