leggedrobotics/raisimLib #27: Feature request: addExternalTorque() with the possibility to define the location of the torque

Atlis, 2019-11-18:
Hi again!

Recently, I had to specify an external torque applied to a specific location on a body. The actual implementation of the `addExternalTorque()` function does not allow to specify the location (and frame in which it is expressed in). Therefore, I had to use `addExternalForce()` to obtain the required effect, but this approach seems sub-optimal from a user standpoint. Would it be possible to overload `addExternalTorque()` to specify the location of the applied torque (in a way similar to `addExternalForce()`)?

Thank you.

On a side note, perhaps the "in-code" documentation could specify where the torque is applied: at the center of mass of the body? or the origin of the body frame? This would clarify a bit. Thanks.

jhwangbo, 2019-11-18:
Hi Atlis. Torque by definition cannot have a location of the application. Torque act on a body but not on a point. Force, on the other hand, has a location.  You cannot apply force to have an equivalent effect since external torque will always result in net-zero force. I know this is confusing. let me know if it is not very clear.

Atlis, 2019-11-19:
Hi jhwangbo. Wow! Ok. I took some time to thoroughly convince myself that it is indeed the fact. It is a shame I never realised such a basic fact earlier. Thank you so much for that one!

closed 2019-11-19 by Atlis