leggedrobotics/raisimLib #28: Crashing of Object function got by raisim::World::getObject only when rendering.

mktk1117, 2019-11-21:
When I have a world and added a height map as below,
```cpp
std::unique_ptr<raisim::World> world;
heightMap = world->addHeightMap(0.0, 0.0, terrainProperties);
heightMap->setName("heightMap");
```
The `getObject` function with name and casting to the `HeightMap` class works fine without rendering.
```cpp
const auto& heightMap = dynamic_cast<HeightMap*>(world->getObject("heightMap"));
double height = heightMap->getHeight(0.0, 0.0);
```
But, when I enable the rendering, it crashes with the same program.

Valgrind gives
>Address 0x4a8 is not stack'd, malloc'd or (recently) free'd
0x53C8FE0: raisim::HeightMap::getHeight(double, double) const (in /opt/raisim/lib/libraisim.so)

jhwangbo, 2019-11-22:
why ```const auto& heightMap``` not ```const auto heightMap```?

jhwangbo, 2019-11-25:
We found that this is an Ogre bug. We posted an issue [here](https://github.com/OGRECave/ogre-assimp/issues/12). Closing this issue for now

closed 2019-11-25 by jhwangbo