leggedrobotics/raisimLib #34: What are the addCompound's parameter?

WatsonZhouAnda, 2019-12-26:
Hi, when I try to use the API "addCompound" in World.hpp to create "Compound" object, I have no idea about its parameter. The reason I want to create this object is that I want to create a "slope" or "wall" or "gap" terrain in order to train in raisimGym.
```
Compound *addCompound(const std::vector<Compound::CompoundObjectChild> &children,
                        double mass,
                        const Mat<3, 3>& inertia,
                        CollisionGroup collisionGroup = 1,
                        CollisionGroup collisionMask = CollisionGroup(-1));
```
Well, I mean what kind of type Compound::CompoundObjectChild  is? I have seen in your Compound.hpp. It used the reference to a list while there is no notation about this "list"
```
struct CompoundObjectChild {
    ObjectType objectType;
    Vec<4> objectParam;
    std::string material;
    Transformation trans;
  };
  Compound(const std::vector<CompoundObjectChild>& list, double mass, Mat<3,3> inertia);
```
So, what "list" should I give? Maybe a urdf path? I have no idea and hope for your reply ASAP.
Thanks in advance.
@jhwangbo @tomlankhorst @vastsoun

jhwangbo, 2019-12-26:
Hi Watson. The documentation is lacking for that method. But for terrain, you can just create multiple static objects. There is no reason to use a compound object.

I'll update the documentation and add an example of a compound object after my break

WatsonZhouAnda, 2019-12-28:
got it, thank you!