leggedrobotics/raisimLib #36: Asking the plan for supporting MacOS

jeeseop, 2020-01-21:
I tried to install raisimLib and raisimOgre in Mac but it didn't work properly.
I could build almost everything but it failed right before finishing the raisimOgre make.
I think that the major reason of the failure is .so files of raisimLib generated for Ubuntu system.
I want to know is there any plan to support MacOS.

Thanks,

jhwangbo, 2020-01-21:
Hi, Jeeseop. The current binaries do not work on Mac because a few memory related calls are linux-only. Mac will be supported in mid this year. Mac support is quite simple but I currently do not have a mac machine to work with...

tomlankhorst, 2020-01-21:
What about running Mac OS on KVM/Qemu, if you have a Mac OS license?

jhwangbo, 2020-01-22:
@tomlankhorst is this how they work on a project for multiple OS'? I have no idea how it is normally done. and unfortunately, I don't have a license

ZQX323, 2020-05-24:
> Hi, Jeeseop. The current binaries do not work on Mac because a few memory related calls are linux-only. Mac will be supported in mid this year. Mac support is quite simple but I currently do not have a mac machine to work with...

Hi Jemin, when will the v1.0 will be published? And if it will support Mac OS? If not, I would love to  develop the Mac OS version.

jhwangbo, 2020-05-24:
V1.0 will be published in June. I already have a working windows version. I will have a version for mac for this release as well.

jeeseop, 2020-06-07:
@jhwangbo I really appreciate for your reply. I look forward to using RaiSim in Mac.

nyxaria, 2020-08-27:
Hi, is the Mac version released? Also, can I ask if it is possible to add prismatic joints with integrated springs (ie series elastic actuator, like a voice coil actuator with an integrated spring)?

jhwangbo, 2020-08-27:
mac version passes all unit tests in my azure dev op. it will be released with v1.0. You can already add a prismatic joint and a spring in the current version. check ```ArticulatedSystem::addSpring```

jhwangbo, 2020-10-29:
Mac version has been pushed to the raisimTech repo. Currently, raisimOgre is not supported but it works with raisimUnity.