leggedrobotics/raisimLib #37: Support for mimic joints

costashatz, 2020-01-27:
Thanks for the nice simulator.

Are there any plans for supporting mimic joints (that URDF supports)? We really need this feature. Thanks a lot! If needed I can provide test URDF files..

jhwangbo, 2020-01-29:
Hi, costashatz. Probably around May. I probably need a full week for this but I dont have that much time at the moment.

costashatz, 2020-01-29:
@jhwangbo thanks for the fast reply.

I will ping you again in May then..

costashatz, 2020-05-02:
@jhwangbo a gentle reminder for making this feature available! Thanks again!

jhwangbo, 2020-05-03:
Yes, this is my next feature to implement. Not sure how long it will take

costashatz, 2020-05-04:
> Yes, this is my next feature to implement.

Great!

> Not sure how long it will take

Ping this issue when you are done (so that I get notified). Thanks!