leggedrobotics/raisimLib #38: How to retrieve contact forces

vastsoun, 2020-01-29:
How can one retrieve contact forces from the simulation?

I have an articulated system and am retrieving contact information using the `robot->getContacts()` member. I tried to compute forces using finite-differences on contact impulses retrieved from `robot->getContacts()[k].getImpulse()` member but the values do not seem to correspond to physical values. I'm simply initializing the robot (a quadruped) in a standing configuration and with the joints frozen via a simple PD joint-space controller.

What would be the most appropriate way of retrieving contact forces from the data afforded by the world of robot instances?

jhwangbo, 2020-01-29:
Hi Vasi, check this link http://raisim.com/sections/Contact.html
I don't understand why you are using finite-difference on contact impulses to retrieve the force. You should simply divide the impulse by the time step to get the force.
(i.e., force * time = impulse)

closed 2020-01-29 by jhwangbo

jacknlliu, 2020-02-19:
@jhwangbo could Raisim get meaningful and continuous contact force data?

jhwangbo, 2020-02-19:
RaiSim can provide contact force as accurate as a physics engine can. There is no non-physical position correction terms which exist in many physics engines. It will be accurate as long as you simulate rigid contacts.

The contact force in RaiSim will not be continuous. Naturally, an impact can lead to a big jump in force. RaiSim also switches computation schemes to minimize the integration error. This can lead to more discontinuous force. But you can turn off this feature by calling 

```cpp
world.getContactSolver().setOrder(true);
```

everytime you call integrate(). This enforces that you use the same solver scheme every time.

I published an article about the contact solver and this is a good reference to understand the contact model of raisim. (https://www.research-collection.ethz.ch/bitstream/handle/20.500.11850/335381/contact-iteration-method%20%284%29.pdf?sequence=1)

jacknlliu, 2020-02-19:
@jhwangbo Will the impact force be as large as the force produced by other physics engines?  Many physics engines generate a very large force about 1e7 which is not reasonable.

jhwangbo, 2020-02-19:
@jacknlliu why do you think this is not reasonable? I can imagine if two billiard balls collide, we can get a very large force. 

If you want a smoother force, you can also run a low-pass filter. Unless you run experiments to accurately measure all ground properties, a low-pass filter will be as good as any other model

jacknlliu, 2020-02-19:
@jhwangbo  thanks for your advice. But I measured the order of magnitude of the force is not consistent with the simulation from some experiments.

jhwangbo, 2020-02-19:
@jacknlliu  If you are measuring a contact force between very hard contacts, the accuracy of the measurement is questionable because your sensor is likely to be more compliant.

As I said, if you want to get an accurate contact force, you have to measure all material properties very accurately. Every collision has its own distinct characteristics. There is no way that a physics engine can provide an accurate result without knowing the details of the materials.

jacknlliu, 2020-02-20:
@jhwangbo  thanks for your tips.! I'll do more research further.

jacknlliu, 2020-02-25:
@jhwangbo I have read the paper about the contact model of raisim, could raisim solve or simulate the jamming contact state of the peg-in-hole task correctly?  The jamming state is a multiple contacts state but more different than other multiple contacts.

jhwangbo, 2020-02-27:
@jacknlliu I haven't tried it but I don't think the peg-in-hole task is going to be a problem.

jacknlliu, 2020-02-27:
@jhwangbo Because the simulation of friction and multiple-contacts are very important in the jamming phenomenon,  many physics engines are often unable to simulate the stable jamming state.

jhwangbo, 2020-02-27:
@jacknlliu you are right. That is why RaiSim uses a successive over-relaxation scheme. So far, I haven't seen a case where it is needed. By default, this feature is turned off (initial alpha is 1). But maybe we need to tune this for jamming conditions. 

I simulated a robot getting jammed in a hole but no tuning was needed. I can't tell how this would change in a perfectly aligned ideal jamming case.