leggedrobotics/raisimLib #40: RaiSim for Gazebo

kumpakri, 2020-02-03:
Hello,

I am using Gazebo simulator for some robot software development and right now I'm stucked on a problem where ODE just seems to be too imprecise to even be used for my case. When I simulate a wheeled robot it slips on the ground no matter how I set the friction, I cannot get it to be realistic. I saw your comparison of common physics engines where the RaiSim rated quite high. I do have certain doubts about the setting of the other physics engines in the tests but knowing that the ODE really does not perform very well for my usecase, I'm wondering if maybe your engine would solve my problems.

I am only a user when it comes to simulation software so I do not understand how it works (and I do not understand the physics you describe in your paper). But say I want to plug the RaiSim physics into Gazebo instead of ODE. Can you describe how would I go about it or point me to any materials that do describe this? When I look into the repos I see no hint on how to do it.

jhwangbo, 2020-02-03:
Hi Kumpakri,

Unfortunately, this is not trivial. I also don't know how to because I am new to Gazebo.
It is on my TODO list but it is not urgent. I will work on it sometime this year.

jacknlliu, 2020-02-19:
@jhwangbo Please consider here [Ignition Gazebo](https://ignitionrobotics.org/docs/citadel/comparison#physics).

> In Ignition Physics, physics engines are integrated as plugins, **so any engine can be integrated without changing the core source code**, as it was the case in Gazebo.

jhwangbo, 2020-02-19:
@jacknlliu I'll definitely check this out when I implement it! Thx

diegoferigo, 2020-02-23:
Hi @jhwangbo, please keep also me in the loop / get in touch if you intend getting closer to Ignition Gazebo. I already have a code skeleton to wrap RaiSim as a Ignition Plugin and quite a lot of experience with the sw architecture of the new simulator. If you recall, when some time ago I opened #24, I was in the middle of early experiments to use RaiSim inside the new Gazebo. Then, I had other priorities and I had to put that side project on pause.

jhwangbo, 2020-02-23:
Hi @diegoferigo, can you share the code?

diegoferigo, 2020-02-23:
@jhwangbo I have a private repository with some code, I can provide you access to have a look if you're interested. Don't expect a working implementation, but it could be a great start :)

Note that at the moment, the upstream version of Ignition Gazebo only supports DART and the name of the shared library of the physics plugin is hardcoded in their Physics system ([some fresh information here](https://bitbucket.org/ignitionrobotics/ign-gazebo/pull-requests/421/support-simulation-with-bullet-physics/diff#comment-133094264)). People are currently working on a new bullet3 backend, but the PR is proceeding quite slow.

In a project I'm developing, I keep a vendored version of the Physics system (the piece of code that hardcodes the plugin name) and in https://github.com/robotology/gym-ignition/pull/83 you can see an initial implementation that prevents using hardcoded names. Though, it's few months old and that PR should be rebased.

I also suggest to have a look to [ign-physics#119](https://bitbucket.org/ignitionrobotics/ign-physics/pull-requests/119/add-simple-example-of-physics-plugin-and/diff) that shows a very contained example.

diegoferigo, 2020-02-25:
Related upstream PR [ignitionrobotics/ign-gazebo#541](https://bitbucket.org/ignitionrobotics/ign-gazebo/pull-requests/541/find-custom-physics-engines).

diegoferigo, 2020-03-05:
Another interesting PR [ignitionrobotics/ign-physics#123](https://bitbucket.org/ignitionrobotics/ign-physics/pull-requests/123/wip-trivial-physics-engine/diff).

diegoferigo, 2020-06-16:
I'm not sure how much you are following upstream's Gazebo efforts, but there have been recently major steps forward towards a more generic and lighter physics plugins implementation:

- [ignitionrobotics/ign-gazebo#541](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-gazebo/pull-requests/541/page/1) (linked also above) enabled actual physics-engine independency
- https://github.com/ignitionrobotics/ign-gazebo/pull/62 drastically reduced the number of features that a physics plugin requires to get integrated into Ignition Gazebo 

[Here's](https://github.com/ignitionrobotics/ign-physics/tree/bullet_chapulina/bullet) a minimal example with the (unmerged, yet) bullet implementation. In light of the benchmark recently done in [costashatz/raisim_dart_benchmarks](https://github.com/costashatz/raisim_dart_benchmarks), I'm looking forward even more to a possible integration in Ignition Gazebo. cc @chapulina

jhwangbo, 2020-06-16:
Frankly, I haven't spent much time on it yet. My focus was on 1. Win/Mac support 2. More unit tests 3. Better unity visualizer 4. performance improvement (~10%) 5. More articulated system methods. 

BTW, Dart and RaiSim use different articulation algorithms. It's not surprising to see such a gap in performance.

I will experiment with the ignition-gazebo next week. 

Before I spend so much time, can you comment on the benefits of the ignition gazebo? why is it better than the original one? and it is popular? how many people use it?

diegoferigo, 2020-06-16:
> My focus was on 1. Win/Mac support 2. More unit tests 3. Better unity visualizer 4. performance improvement (~10%) 5. More articulated system methods.

Cool, good to know. 10% of boost is a great achievement considering the good current performance.

> I will experiment with the ignition-gazebo next week.

My comment's purpose was providing a quick update to the status and link resources together :) 

In my opinion creating a Ignition Physics is not a task that could be done in one day. There's [some documentation](https://github.com/ignitionrobotics/ign-physics/tree/ign-physics2/tutorials) and much could be adapted from other existing backends, but the development requires some domain knowledge, in particular:

- How the SDF are parsed and stored in memory as Entities and Components, done by [`SdfEntityCreator`](https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/SdfEntityCreator.cc) (URDFs are converted to SDF by gazebo)
- How these Entities and Components are re-assembled as SDF and passed to the physics plugin [by the Physics system](https://github.com/ignitionrobotics/ign-gazebo/blob/1c451db83e384411d5fd739ac02b4303fab3773c/src/systems/physics/Physics.cc#L599-L1033)
- When the physics plugin will receive these chunks, raisim should create link-by-link and joint-by-joint the model (maybe possible after #24)
- Finally, the physics plugin should implement the Features as described in the documentation and return the right data type to Gazebo.

All the rest is handled by gazebo: GUI, rendering (with multiple backends, ogre ogre2 and partially optix), distributed simulations, and so on.

> Before I spend so much time, can you comment on the benefits of the ignition gazebo? why is it better than the original one? and it is popular? how many people use it?

Probably the developers can explain it better than me, but the old Gazebo Classic will receive only bug fixing starting from the current v11. Ignition Gazebo is currently used for the new [Darpa SubT challenge](https://subtchallenge.com/) and as far as I understood most of the recent development effort went into this new version. The new simulator has been rewritten in favour of modularity and abstraction (physics and rendering engines are plugins that can be switched very easily).

Personal opinion: Gazebo Classic has a huge user base that will progressively migrate towards Ignition Gazebo. In my lab we embraced it since a while now, and I'm personally very happy about it. Especially for RL experiments, the new simulator-as-a-library feature (similar to raisim, btw) is great. Before, due to its previous architecture, it was impossible to integrate Gazebo Classic in such big and distributed experiments.

chapulina, 2020-06-16:
> creating a Ignition Physics plugin is not a task that could be done in one day. 

I agree that there's a learning curve. I'd also mention that ideally someone integrating a new engine through `ign-physics` wouldn't need to know anything about `ign-gazebo`'s internals. [This tutorial](https://github.com/ignitionrobotics/ign-physics/blob/ign-physics2/tutorials/03_physics_plugins.md) has some helpful info.

> the old Gazebo Classic will receive only bug fixing starting from the current v11

Correct

> Gazebo Classic has a huge user base that will progressively migrate towards Ignition Gazebo.

That's the hope :crossed_fingers: 

> can you comment on the benefits of the ignition gazebo?

I think @diegoferigo made a great case for it. For those interested in plugging in their own physics engine, it's by all means better than Gazebo-classic.