leggedrobotics/raisimLib #41: Asking for the constraints

jeeseop, 2020-02-13:
I saw that there is the stiffwire constraints
Is there any plan to add distance only constraints? like rod with ball joint on each end.

jhwangbo, 2020-02-13:
If it is an equality constraint, you can do it with an articulated system. You have to write a URDF file for that but it is not very complex.

jeeseop, 2020-02-14:
Here is the followup question.
Is it possible to attach urdf model and raisim defined body using some specific joint setup?

jhwangbo, 2020-02-15:
You can create a child class (an example in raisimOgre) to build your own articulated system in cpp. You can attach an existing body to a child class and create a new articulated system. This is cumbersome but does exactly what you described. I have a plan to simplify this soon

What would be the use case of this feature? Can you describe what you want to do?

closed 2020-02-16 by jhwangbo