leggedrobotics/raisimLib #48: Generalized Coordinate Questions (and conversion from world frame to body frame)

Neotriple, 2020-03-26:
Hi there,

A question regarding the convention you're using for the generalized coordinates:

Is the quaternion in the convention: x, y, z, w
or: w, x, y, z?

Similarly, what is the convention you're using for the generalized velocity vector regarding the angular velocity? Is this in XYZ Euler, 

Finally, I'm trying to get both the generalizedCoordinates and generalizedVelocities from the world frame to the body frame. There doesn't seem to be a function for this, is that correct? If so, would you have any suggestions on converting these to their respective body coordinate frames?

jhwangbo, 2020-03-26:
1. it is w, x, y, z
2. the generalized velocity of a floating body is v_x, v_y, v_z, w_x, w_y, w_z, all of which are expressed in the world frame. Note that most simulators express the angular velocity (i.e., w) in the body frame. RaiSim uses the World frame to speed up the simulation
3. all Getters starts with "get" prefix. So to get the generalizedCoordinate, you use a method "getGeneralizedCoordinate()". same for the generalized velocity. If you want to get both at once, you can also use getState() method.

Neotriple, 2020-03-26:
Hi Jemin,

Thanks for the response.

Regarding #3, I should have been more specific. As you mentioned, I can get the generalized Coordinate and Velocities using the getters. However, as you mentioned in the documents (and in your comment) that everything in raisim is done in world frame. My controller is all done in local generalized coordinates, so I'm just trying to see how I can take the results from simulation in raisim and convert them to local coordinates. 

Is it correct to just multiply the world frame translational and angular velocities by the rotation matrix that maps from the world to body (i.e. can i just do v_b = R_bw * v_w and similarly for the w terms)? In this case, v_b is the translational velocity vector in base frame, R_bw is the rotational matrix from world to base frame, and v_w is the translational velocity vector in world frame (which I can get from the methods you have provided).

I believe I can use the method getBaseOrientation() which should return the rot_WB matrix which I can then transpose to get R_bw, or is there something I'm missing?

jhwangbo, 2020-03-26:
You are correct. You are not missing anything as far as I see

closed 2020-03-27 by jhwangbo