leggedrobotics/raisimLib #51: Segmentation fault for articulated system with 0 movable joints

wxmerkt, 2020-04-21:
While this may sound like a contradiction, it's perhaps useful to debug a URDF that only contains fixed or no joints at all (root link only).

**To reproduce:** call `addArticulatedSystem` on a URDF with a base link and no further joints or links.

**Observed behaviour:** Segmentation fault.

**Expected behaviour:** Adds the system with no movable joint (single rigid body) _or_ throws an exception saying that at least one movable (revolute, continuous, prismatic, spherical) joint is required.

jhwangbo, 2020-04-21:
I know this does not work in raisim. I doubt if this can be useful but I'll fix this in v1.0.

Does your URDF have no movable joints?

wxmerkt, 2020-04-21:
No, this happened when trying to work on a minimum working example for #50. I have seen URDFs with no movable joints though (some people use them to load collision environments, e.g., tables or thelike).

jhwangbo, 2020-04-21:
This will give me a segfault for a different reason. Such a system is not supported in RaiSim. 

It will be supported in v1.0. It is going to be an easy fix. I just couldn't anticipate that all fixed systems can be useful for some people...

wxmerkt, 2020-04-21:
Okay - I was more keen to have an exception thrown or error shown than a segmentation fault :-).

closed 2020-04-21 by wxmerkt