leggedrobotics/raisimLib #52: Robot from SolidWorks-sourced URDF falling through floor

dyspr0sium, 2020-04-29:
I am writing a Gym environment for a robot in Python using raisimpy, following the ANYmal example environment from [raisimpy's examples](https://github.com/robotlearn/raisimpy/blob/master/examples/raisimpy_gym/envs/anymal/env.py). The ANYmal URDF loads in fine but using a URDF exported from SolidWorks, I cannot see the robot on the floor. With the floor not loaded visually in raisimOgre, I can see the robot falling down to infinity. I suspect there is a problem with the collision properties in the URDF, which are linked to OBJ files exported from SolidWorks rather than simple shapes defined directly in the URDF. The URDF is as follows:
```
<?xml version="1.0" encoding="utf-8"?>
<!-- This URDF was automatically created by SolidWorks to URDF Exporter! Originally created by Stephen Brawner (brawner@gmail.com) 
     Commit Version: 1.5.1-0-g916b5db  Build Version: 1.5.7152.31018
     For more information, please see http://wiki.ros.org/sw_urdf_exporter -->
<robot
  name="Quadrupedal Robot URDF ready.SLDASM">
  <link
    name="Body">
    <inertial>
      <origin
        xyz="6.3895E-11 0.00012763 -0.017744"
        rpy="0 0 0" />
      <mass
        value="0.52729" />
      <inertia
        ixx="0.0019565"
        ixy="-6.0032E-10"
        ixz="1.2826E-08"
        iyy="0.00076333"
        iyz="-4.6881E-06"
        izz="0.0025305" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="../meshes/Body.stl.obj" />
      </geometry>
      <material
        name="">
        <color
          rgba="0.75294 0.75294 0.75294 1" />
      </material>
    </visual>
    <collision>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="../meshes/Body.stl.obj" />
      </geometry>
    </collision>
  </link>
  <link
    name="L11">
    <inertial>
      <origin
        xyz="0.0243995568220635 -0.0139117596798611 -0.0357904787858385"
        rpy="0 0 0" />
      <mass
        value="0.0542977426736306" />
      <inertia
        ixx="1.67959891588832E-05"
        ixy="3.99700363772074E-07"
        ixz="4.4933479303535E-07"
        iyy="2.2247186909432E-05"
        iyz="-1.73903214218687E-06"
        izz="1.69288569652552E-05" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
[... 865 lines cut ...]
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="../meshes/L44.stl.obj" />
      </geometry>
    </collision>
  </link>
  <joint
    name="Foot4"
    type="fixed">
    <origin
      xyz="-0.0062871 0.0033624 0.095646"
      rpy="1.5708 0 1.5708" />
    <parent
      link="L43" />
    <child
      link="L44" />
    <axis
      xyz="0 0 1" />
  </joint>
</robot>
```

I can provide the collision OBJs if necessary. Appreciate any help.

jhwangbo, 2020-04-29:
I can't tell much without checking out the mesh files.

dyspr0sium, 2020-04-30:
Sure, here are the [meshes](https://github.com/leggedrobotics/raisimLib/files/4555922/meshes.zip). The original files from SolidWorks are STL but I converted them to OBJ because there was an error importing the original STL meshes into raisim.

I should add that the URDF, regardless of STL or OBJ meshes, was checked to work correctly in PyBullet.

jhwangbo, 2020-04-30:
well, apparently I was wrong about the obj convention. You obj file has a line like
```v  -0.00381818 0.000196837 -0.00616337```
There are two spaces after the "v". same for "f". My obj reader can only read if there is only one space...
I fixed this is in V1.0 but you can't get this version yet. I suggest just replace all double spaces with a single space. Thank you for reporting this bug

and your meshes seem very fine. This creates so much burden on collision detection. You will lose the benefits of the super-fast dynamics solver of RaiSim with this model...

dyspr0sium, 2020-04-30:
Perfect, thank you very much. The issue is resolved after replacing all double spaces with single spaces in the OBJ files.

I will consider replacing the collision boxes with simple shapes if the performance too slow due to the fine meshes.

closed 2020-04-30 by dyspr0sium