leggedrobotics/raisimLib #54: Calling getLinearVelocity() on a kinematic object

christophecricket, 2020-06-08:
Hi There
 I have a simulation where I am imposing a position on a kinematic object, and trying to read its velocity with getLinearVelocity(), but the output I get is 0. Is this something I'm doing wrong on my side, or is this expected behaviour?

The same method seems to work fine for an object I have set to dynamic.

jhwangbo, 2020-06-08:
I just tested the code and it works for me. For kinematic objects, the velocities don't change. So you set them to certain numbers and they remain the same. You can assign both linear and angular velocities with `setVelocity`. If you didn't set them, they are zero by default.

christophecricket, 2020-06-09:
Aha, gotcha. I thought ```World.integrate()``` might differentiate the position command as part of the engine's inner workings. Good to know, I guess I'll compute/read velocity some other way, then. Thanks!

closed 2020-06-09 by christophecricket