leggedrobotics/raisimLib #56: about the getContactPointVel method in ArticulatedSystem

WSPeng, 2020-06-24:
Hi, 

I am trying to use the getContactPointVel method, with the pointID and raisim::Vec<3> vel as arguments. but I get segfault. Do you have any suggestion to fix the problem? 

for example, then code is: 
raisim::Vec<3> footVel_hl;
footVel_hl.setZero(); 
for(auto& contact: robot -> getContacts()) {
  if ( footIndex_hl == contact.getlocalBodyIndex() ) {
    robot->getContactPointVel(footIndex_hl, vec3_);
}
}

jhwangbo, 2020-06-24:
getContactPointVel takes the contactId as the first argument. There can be many contacts on body "foddIndex_hl" so your syntax cannot be used. So you should do

`raisim::Vec<3> footVel_hl;
footVel_hl.setZero();
for(size_t i=0; i<robot->getContacts().size(); i++) {
if ( footIndex_hl == robot->getContacts()[i].getlocalBodyIndex() ) {
robot->getContactPointVel(i, footVel_hl);
}
}`

Not tested but should work. Let me know if it doesn't

WSPeng, 2020-06-24:
Yes, it works! Thank you ;D

closed 2020-06-24 by WSPeng