leggedrobotics/raisimLib #58: Asking for how to set static object

ZQX323, 2020-07-05:
Hi,

I want to add some small spheres in the space as reference motion. So I want them hang in the space without falling. I am struggling with this. Any suggestions? Besides, I want them not collide with the robot. After I set CollisionGroup and CollisionMask, it does not work, so I am a little confused to use CollisionMask. How to set it? Thank you very much!

jhwangbo, 2020-07-05:
sphere->setBodyType(raisim::BodyType::STATIC);
But this will still allow collisions.

I didn't make documentation for collision filtering but it is standard in all physics engines. Check this Link
https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=1086

closed 2020-07-05 by jhwangbo

ZQX323, 2020-07-05:
Thanks very much, it really helps!