leggedrobotics/raisimLib #59: raisim integration scheme

christophecricket, 2020-07-08:
Hi

I can't find it in the documentation, and the source is difficult to peruse, but what kind of an integration scheme does raisim use? I'm assuming it's some kind of a fixed step explicit method, but I was just wondering which one.

jhwangbo, 2020-07-08:
PD controller and passive elements (springs and dampers) are integrated using an implicit method (http://raisim.com/sections/ArticulatedSystem.html#pd-controller). The dynamics and contacts are integrated using a time-stepping method (https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=8255551). The position is updated using a trapezoidal rule (which I found to be more accurate than explicit and semi-implicit methods.). 

I think this is too much detail for users. Why do you want to know about the integration scheme? How would this be useful for the users?

christophecricket, 2020-07-09:
I'm using Raisim to render force for a teleoperation system (shown in part 2 of this, if you're interested https://ieeexplore.ieee.org/document/88057 ). I was having some problems last week with the quality of my rendered force, and was wondering if solver errors were playing a part. Turns out this wasn't the case, but I thought I'd ask so that I know next time it pops up.

I think you probably have a better idea for the use cases of Raisim than I do, so you're probably a much better judge of whether this would be useful for users than I would be. If I'm the first person to ask this, maybe it shouldn't be your highest priority, indeed :)

Just out of interest, I thought the trapezoidal rule counted as a semi-implicit integration method. What's the distinction?

jhwangbo, 2020-07-29:
I guess I forgot to answer this. Trapezoidal rule is x_{t+1} = x_{t} + (v_{t} +v_{t+1}) * 0.5 * dt. semi implicit is x_{t+1} = x_{t} + v_{t+1} * dt

closed 2020-07-29 by jhwangbo