leggedrobotics/raisimLib #60: How to implement a HeightMap properly

christophecricket, 2020-07-15:
Hi Again

I'm trying to implement a HeightMap for the "ground" of my simulation with the following code. It's supposed to be 10*10m wide with a spacing in x and y of 0.1m.

```
//heightmap parameters
int xSamples = 101;
int ySamples = 101;
double xScale = 10;
double yScale = 10;
std::vector<double> height_map = {0};
height_map.resize(10201);
double heightmap_row_number = 0;
double heightmap_col_number = 0;

for(int i=0; i < 10201; ++i){
    if(heightmap_row_number == 100){
    	++heightmap_col_number;
    	heightmap_row_number = 0;
    }

    height_map.at(i) = -10 + heightmap_col_number*0.1 + heightmap_row_number*0.1;

    ++heightmap_row_number;
}
raisim::HeightMap* ground = world.addHeightMap(xSamples,ySamples,xScale,yScale,0,0,height_map,
 											"default",1,CollisionGroup(-1));
```

However, in the while loop where I call ```world.integrate()```, the call fails on the third iteration, throwing a ```std::bad_array_new_length``` error. What am I doing wrong?

jhwangbo, 2020-07-15:
have you tried to visualize the map before integration? You can set the integration stepping manual and check it in raisimOgre. The code looks to me. Maybe it's not the height map that fails.

christophecricket, 2020-07-15:
I've been using ROS visualisation tools for my visualisation so far. Before I get started with raisimOgre, could you maybe tell me how the height vector transforms into the height map? Do row and column numbers map directly to x and y values? Or is it packed in a different manner?

jhwangbo, 2020-07-15:
[image]

Copy&pasting your code looks like this.

The height map fills (x0, y0), (x1, y0), (x2, y0), and so on.

christophecricket, 2020-07-15:
Just to clarify, the sampling on the (x,y) plane for the heightmap then looks like this, right? (If xSamples=ySamples=3)

[
[image]
](url)

christophecricket, 2020-07-15:
The heightmap as I've defined it in my code is supposed to represent a flat plane. Is that what you see in raisim ogre? Because if not then it's a mistake.

jhwangbo, 2020-07-15:
using if(heightmap_row_number == 101), I get this
[image]

very stiff but flat.

In your example, using heightMapVector = {z0, z1, ..., z8}, we get
(x0, y0, z0), (x1, y0, z1), (x2, y0, z2), (x0, y1, z3), ...

christophecricket, 2020-07-15:
Aww man... Yeah that works. No more errors. I guess my off-by-one error made an overly discontinuous height map and I guess making a meshed surface from that creates some sort of an error? 

Sorry for wasting your time on my stupid math mistakes, though.

jhwangbo, 2020-07-15:
Having a very discontinuous terrain is fine. I think your robot was probably buried inside the terrain. Such an unrealistic scenario can lead to unrealistic simulation.

christophecricket, 2020-07-15:
Gotcha. Thanks for the help!

closed 2020-07-15 by christophecricket