leggedrobotics/raisimLib #62: Question on the materials used in function addHeightMap.

Fleer-code, 2020-08-17:
Hello, I wonder if there is a way to change the coefficient of friction between the robot and the ground. I have found a parameter `material` in function
 `  HeightMap *addHeightMap(const std::string &raisimHeightMapFileName,
                          double centerX,
                          double centerY,
                          const std::string &material = "default",
                          CollisionGroup collisionGroup = 1,
                          CollisionGroup collisionMask = CollisionGroup(-1));`.
Is there any other choice to change the "default" into other materials? 
I have also found a file, _raisim_workspace/raisimLib/examples/rsc/basicMaterials.xml_ which has definition of friction and restitution. Is it related to the the parameter `material` in addHeightMap? If yes, how should I modify or insert material parameters in this file, and how can I adopt these materials in my own code file?

jhwangbo, 2020-08-17:
Check this link. http://raisim.com/sections/MaterialSystem.html

closed 2020-08-17 by jhwangbo

Fleer-code, 2020-08-18:
Thanks. That's exactly what I need.

Fleer-code, 2020-08-18:
There is another question I feel confused. Is there a way to change the material of the links of the robot, which is loaded from .urdf file. I found a function ` void setMaterial(const std::string &material);` in JointAndBodies.hpp, but haven't figure out how to use it.

jhwangbo, 2020-08-18:
articulated system materials are specified differently. I added description here http://raisim.com/sections/MaterialSystem.html#example-articulated-systems

Fleer-code, 2020-08-19:
Thanks. That's very helpful!